RyoKosaka/HelloDrum-arduino-Library #6: Do you have an ideas about Position Sensing?

mrunzo, 2019-10-18:
I would like to implement positional sensing feature if you have any idea you can share an idea and resources.

thijstriemstra, 2020-03-11:
Would love to see this feature as well.

corrados, 2020-12-12:
Recently, I started researching the signal processing of a drum trigger module. As a first step I wanted to create a proof-of-conecpt where I ignore the latency and the processing requirements of the algorithms.

First results are shown here: https://github.com/corrados/mypitools/blob/master/projects/edrumtrigger/README.md.

I am not sure if this will lead to any useful algortihm which could be implemented on an Ardunio. But it is fun to do some research in that area :-).

corrados, 2020-12-13:
I have now finished a first working version and created a short Youtube video to show my algorithms in action: https://youtu.be/6eQjCD-DFjo

corrados, 2020-12-20:
I have ported the code to an ESP32 microcontroller now. The code is available at: https://github.com/corrados/edrumulus.
Here is a short video showing the trigger performance of the current implementation: https://youtu.be/naP-ODXl9Y0.

thijstriemstra, 2020-12-21:
@corrados wow that's awesome! Have you tried it with any VST, eg. Superior Drummer? I will order a new ESP32 and make this a holiday project.

corrados, 2020-12-21:
No, I have not done that yet. I have just used the Roland TDW-20 to generate the drum sound from the MIDI signal. If you want to try out my library, please keep in mind that the code is in a very early stage and only optimized for my PD-120 mesh pad. Also, you have to put the operating point of the ADC in the middle of its range (I do it with two resistors in series connection and a capacitor).

thijstriemstra, 2020-12-21:
> the code is in a very early stage and only optimized for my PD-120 mesh pad. 

I am also using one of those PD-120 mesh pads 👍🏼

SunboX, 2022-04-23:
Hey @mrunzo , I had the same in my mind. See issue #43 . I wrote the issue, before I saw yours.

ojungleboy, 2022-04-27:
Very interested on this!