RyoKosaka/HelloDrum-arduino-Library #10: ATmega328P Maximum Open e-Drum Kit

hori64, 2020-01-14:
I would like to thank first of all this shared project. I also built my own kit. ESP32 hasn't arrived yet, so I tried an Arduino pro mini module. After memory optimization this was successful.

HelloDrum kick(0);
HelloDrum snare(1,2);
HelloDrum tom_1(3);
HelloDrum tom_2(4);
HelloDrum tom_3(5);
HelloDrum tom_4(A0);
HelloDrum hihat(6);
HelloDrum hihatControl(7);
HelloDrum cym_1(A1);
HelloDrum cym_2(A2);
HelloDrum ride(A3,A6);

This is an "ATmega328P Maximum Open e-Drum Kit" :)

I will send the necessary changes in 2 parts.
I would appreciate it if they were included in the official library.
THX!

jhsa, 2020-12-23:
Hi, I know that you posted nearly one year ago but I would really like to know how you managed to connect so many pads to the Pro Mini. I am using a Nano with MUX (4067), and after 8 pads I am using all memory :( 
Thanks.

RyoKosaka, 2020-12-24:
his code is here
https://github.com/hori64/Open-e-Drum-Kit

hori64, 2020-12-24:
Hi, Yes, I only used the drum last year, I didn’t improve it further.I just tried it with the latest 0.7.7 library version and with that I also have low memory for global variables. (112%)Version 0.7.4 is no longer good for me either. (101%)Version 0.7.3 is still perfect for me. (97%) Hori Feladó: Ryo KosakaElküldve: 2020. december 24., csütörtök 8:39Címzett: RyoKosaka/HelloDrum-arduino-LibraryMásolatot kap: hori64; AuthorTárgy: Re: [RyoKosaka/HelloDrum-arduino-Library] ATmega328P Maximum Open e-Drum Kit (#10) his code is herehttps://github.com/hori64/Open-e-Drum-Kit

jhsa, 2020-12-24:
Thanks guys..
By the way, there is a Facebook group which could be used to discuss the helloDrum / Open E-Drums project.
You are very welcome to join it. here is the link:

https://www.facebook.com/groups/838951150235243