RyoKosaka/HelloDrum-arduino-Library #17: All pads sound like snare

soul006, 2020-04-03:
Hi everyone.  The module plays all the pads as the snare.  I am using Arduino mega R3, and code 0.7.3.  Do you have any idea what it could be?  All modules sound but the snare is always signaled in ezdrummer.  I am in the last stage of the project, it has cost me a lot since I do not understand English well, I feel a little frustrated.  Greetings from Argentina!

RyoKosaka, 2020-04-19:
@soul006 
Can you show me the code you're using?
Do you have a schematic?

My guess is that if you're using EEPROM, the initial value of the note number is 38, so it's a snare sound.

soul006, 2020-04-20:
#include <hellodrum.h>
#include <MIDI.h>
#include <LiquidCrystal.h>

MIDI_CREATE_DEFAULT_INSTANCE();

//LCD pin define
LiquidCrystal lcd(12, 11, 5, 4, 3, 2); //(rs, en, d4, d5, d6, d7)

//Please name your pad and controller.
HelloDrum kick(0);
HelloDrum snare(1);
HelloDrum rimshot(2);
HelloDrum hihat(3);
HelloDrum hihatControl(4);
HelloDrum toma(5);
HelloDrum tomb(6);
HelloDrum tomc(7);
HelloDrum crash(8);
HelloDrum ride(9);

//Set the DIGITAL pin number to which the buttons are connected.
HelloDrumButton button(6, 7, 8, 9, 10); //(EDIT,UP,DOWN,NEXT,BACK)

void setup()
{
  //if you use ESP32, you have to uncomment the next line.
  //EEPROM_ESP.begin(512);

  //If you use Hairless MIDI, you have to comment out the next line.
  MIDI.begin(10);
  //And uncomment the next two lines. Please set the baud rate of Hairless to 38400.
  //MIDI.begin();
  //Serial.begin(38400);

  //Give each pad a name to be displayed on the LCD.
  //It is necessary to make the order in exactly the same order as you named the pad first.
  
  kick.settingName("KICK");
  snare.settingName("SNARE");
  rimshot.settingName("RIMSHOT");
  hihat.settingName("HIHAT");
  hihatControl.settingName("HIHAT PEDAL");
  toma.settingName("TOM A");
  tomb.settingName("TOM B");
  tomc.settingName("TOM C");
  crash.settingName("CRASH");
  ride.settingName("RIDE");

  //Load settings from EEPROM.
  //It is necessary to make the order in exactly the same order as you named the pad first.
  kick.loadMemory();
  snare.loadMemory();
  rimshot.loadMemory();
  hihat.loadMemory();
  hihatControl.loadMemory();
  toma.loadMemory();
  tomb.loadMemory();
  tomc.loadMemory();
  crash.loadMemory();
  ride.loadMemory();

  //boot message
  lcd.begin(16, 2);
  lcd.clear();
  lcd.print("MIDI Drum");
  lcd.setCursor(0, 1);
  lcd.print("Iniciando...");
}

void loop()
{

  /////////// 1. LCD & SETTING MODE /////////////

  bool buttonPush = button.GetPushState();
  bool editStart = button.GetEditState();
  bool editDone = button.GetEditdoneState();
  bool display = button.GetDisplayState();

  char *padName = button.GetPadName();
  char *item = button.GetSettingItem();
  int settingValue = button.GetSettingValue();

  button.readButtonState();

  kick.settingEnable();
  snare.settingEnable();
  rimshot.settingEnable();
  hihat.settingEnable();
  hihatControl.settingEnable();
  toma.settingEnable();
  tomb.settingEnable();
  tomc.settingEnable();
  crash.settingEnable();
  ride.settingEnable();

  if (buttonPush == true)
  {
    lcd.clear();
    lcd.print(padName);
    lcd.setCursor(0, 1);
    lcd.print(item);
    lcd.setCursor(13, 1);
    lcd.print(settingValue);
  }

  if (editStart == true)
  {
    lcd.clear();
    lcd.print("EDITAR");
    delay(500);
    lcd.clear();
    lcd.print(padName);
    lcd.setCursor(0, 1);
    lcd.print(item);
    lcd.setCursor(13, 1);
    lcd.print(settingValue);
  }

  if (editDone == true)
  {
    lcd.clear();
    lcd.print("COMPLETO");
    delay(500);
    lcd.clear();
    lcd.print(padName);
    lcd.setCursor(0, 1);
    lcd.print(item);
    lcd.setCursor(13, 1);
    lcd.print(settingValue);
  }

  //show hitted pad name and velocity to LCD
  if (display == true)
  {
    int velocity = button.GetVelocity();
    char *hitPad = button.GetHitPad();

    lcd.clear();
    lcd.print(hitPad);
    lcd.setCursor(0, 1);
    lcd.print(velocity);
  }

  ////////// 2. SENSING & SENDING MIDI////////////

  //Sensing each pad.
  kick.singlePiezo();
  snare.singlePiezo();
  rimshot.singlePiezo();
  hihat.HH();
  hihatControl.TCRT5000();
  toma.singlePiezo();
  tomb.singlePiezo();
  tomc.singlePiezo();
  crash.singlePiezo();
  ride.singlePiezo();

  //Sending MIDI signals.
  //KICK//
  if (kick.hit == true)
  {
    MIDI.sendNoteOn(kick.note, kick.velocity, 10); //(note, velocity, channel)
    MIDI.sendNoteOff(kick.note, 0, 10);
  }

  //SNARE//
  if (snare.hit == true)
  {
    MIDI.sendNoteOn(snare.note, snare.velocity, 10); //(note, velocity, channel)
    MIDI.sendNoteOff(snare.note, 0, 10);
  }

  //RIIMSHOT//
  if (rimshot.hit == true)
  {
    MIDI.sendNoteOn(rimshot.note, rimshot.velocity, 10); //(note, velocity, channel)
    MIDI.sendNoteOff(rimshot.note, 0, 10);
  }

  //HIHAT//
  if (hihat.hit == true)
  {
    //check open or close
    //1.open
    if (hihatControl.openHH == true)
    {
      MIDI.sendNoteOn(hihat.noteOpen, hihat.velocity, 10); //(note of open, velocity, channel)
      MIDI.sendNoteOff(hihat.noteOpen, 0, 10);
    }
    //2.close
    else
    {
      MIDI.sendNoteOn(hihat.noteClose, hihat.velocity, 10); //(note of close, velocity, channel)
      MIDI.sendNoteOff(hihat.noteClose, 0, 10);
    }
  }

  //HIHAT CONTROLLER//
  //when hihat is closed
  if (hihatControl.closeHH == true)
  {
    MIDI.sendNoteOn(hihatControl.note, hihatControl.velocity, 10); //(note of pedal, velocity, channel)
    MIDI.sendNoteOff(hihatControl.note, 0, 10);
  }
  //sending state of pedal with controll change
  if (hihatControl.moving == true)
  {
    MIDI.sendControlChange(4, hihatControl.pedalCC, 10);
  }

  //TOM A//
  if (toma.hit == true)
  {
    MIDI.sendNoteOn(toma.note, toma.velocity, 10); //(note, velocity, channel)
    MIDI.sendNoteOff(toma.note, 0, 10);
  }
  
  //TOM B//
  if (tomb.hit == true)
  {
    MIDI.sendNoteOn(tomb.note, tomb.velocity, 10); //(note, velocity, channel)
    MIDI.sendNoteOff(tomb.note, 0, 10);
  }
  
  //TOM C//
  if (tomc.hit == true)
  {
    MIDI.sendNoteOn(tomc.note, tomc.velocity, 10); //(note, velocity, channel)
    MIDI.sendNoteOff(tomc.note, 0, 10);
  }
  
  //CRASH//
  if (crash.hit == true)
  {
    MIDI.sendNoteOn(crash.note, crash.velocity, 10); //(note, velocity, channel)
    MIDI.sendNoteOff(crash.note, 0, 10);
  }

  //RIDE//
  if (ride.hit == true)
  {
    MIDI.sendNoteOn(ride.note, ride.velocity, 10); //(note, velocity, channel)
    MIDI.sendNoteOff(ride.note, 0, 10);
  }
}

RyoKosaka, 2020-04-20:
@soul006 
I can't find any problems with your code.

Is there a piezo or sensor connected to all the analog pins that are declared for use in the code?
If it's not connected, then the pin is floating. Then the analog pin will receive a random value, and it will be mistaken for a hit.
I do not yet have a solution to this.
All you have to do is get rid of the floating analog pins.

soul006, 2020-04-21:
I'm going to check just like you say.  On the other hand I consult you for the steps I am doing with EEPROM: I load the EEPROM program, I load the HelloDrum program, finally I convert the board with hyduino.  
I am also thinking of resorting to version 0.7 which at one point had given me results.  
From already thank you very much for your time.

NinjaPanas, 2020-11-03:
Well, I kind of remember that when I started the device for the first time (version 0.7.6, arduino UNO, firmware for dual MOCO), all the "points" were tuned to one midi tone. I manually tuned each drum to the tone I needed, and after turning the unit off and on again, all the saved settings were in place.

But if I hadn't manually tuned the tones, then all the drums would have sounded with some one tone. Maybe it could be a snare drum too.