RyoKosaka/HelloDrum-arduino-Library #19: HELP use 31 pads*4MUX but the flashing LCD shows SPLASH1. Unable to press any buttons

attachi201, 2020-06-28:
Hi i use 31 pads but the flashing LCD shows SPLASH1.
can't press any buttons,
but when i comment code scan mux So it works  
i use Mega2560 with 4*MUX IC and LCDI2C
this is my code 

`
#include <hellodrum.h>
#include <MIDI.h>
#include <LiquidCrystal_I2C.h>

LiquidCrystal_I2C lcd(0x27, 16, 2);

MIDI_CREATE_DEFAULT_INSTANCE();

/*
Mux Breakout ----------- Arduino
S0 ------------------- 2
S1 ------------------- 3
S2 ------------------- 4
Z -------------------- A0
 */
//Define MUX Pins

HelloDrumMUX_4051 mux1(2,3,4,0); //D2, D3, D4, A0

HelloDrumMUX_4051 mux2(5,6,7,1); //D5, D6, D7, A1

HelloDrumMUX_4051 mux3(8,9,10,2); //D8, D9, D10, A2

HelloDrumMUX_4051 mux4(11,12,13,3); //D11, D12, D13, A2

//Please name your pad and controller.

HelloDrum snare(0, 1);
HelloDrum hihat(2, 3);
HelloDrum ride(4, 5);
HelloDrum kick1(6);
HelloDrum kick2(7);
HelloDrum tom1(8);
HelloDrum rimtom1(9);
HelloDrum tom2(10);
HelloDrum rimtom2(11);
HelloDrum tom3(12);
HelloDrum rimtom3(13);
HelloDrum tom4(14);
HelloDrum rimtom4(15);
HelloDrum tom5(16);
HelloDrum rimtom5(17);
HelloDrum crash1(18, 19);
HelloDrum crash2(20, 21);
HelloDrum crash3(22, 24);
HelloDrum crash4(25, 26);
HelloDrum china1(27);
HelloDrum china2(28);
HelloDrum splash1(29);
HelloDrum splash2(30);
HelloDrum hihatPedal(23);

//Set the DIGITAL pin number to which the buttons are connected.
HelloDrumButton button(14, 15, 16, 17, 18); //(EDIT,UP,DOWN,NEXT,BACK)

void setup()
{
  lcd.init(); // initialize the lcd
  lcd.backlight();

  //if you use ESP32, you have to uncomment the next line.
  //EEPROM_ESP.begin(512);

[... 505 lines cut ...]
  if (crash4.hit == true)
  {
    MIDI.sendNoteOn(crash4.note, crash4.velocity, 10); //(note, velocity, channel)
    MIDI.sendNoteOff(crash4.note, 0, 10);
  }

  //edge
  else if (crash4.hitRim == true)
  {
    MIDI.sendNoteOn(crash4.noteRim, crash4.velocity, 10); //(note, velocity, channel)
    MIDI.sendNoteOff(crash4.noteRim, 0, 10);
  }

  //choke
  if(crash4.choke == true)
   {
  
     MIDI.sendPolyPressure(crash4.note, 127, 10);
    MIDI.sendPolyPressure(crash4.noteRim, 127, 10);
    MIDI.sendPolyPressure(crash4.note, 0, 10);
    MIDI.sendPolyPressure(crash4.noteRim, 0, 10);
   }
  

}`

jhsa, 2020-11-27:
Hi, did you fix the problem? Thanks

RyoKosaka, 2020-11-27:
If there is nothing connected to the analog or MUX pins for sensing (floating pins), it will pick up noise and be perceived as a continuous hit. Connect a pad to all the pins, or if not, use a 1MΩ resistor to connect them to GND.

attachai201, 2021-05-04:
You're right, the code can now be used, thank you very much.

attachai201, 2021-05-04:
@jhsa Now it works. But you have to connect all pads to the board.