RyoKosaka/HelloDrum-arduino-Library #26: Circuit for TCRT5000

NinjaPanas, 2020-09-17:
I redrawn the circuitry and found that the 4.7K resistor could be positioned like this:
[image]

I don't understand why this resistor is there and can I replace two 10K and 4.7k resistors with one 3.3K resistor?

Is there a special connection diagram for the TCRT5000 for MUX?

NinjaPanas, 2020-09-18:
I assembled a device for 5 piezo and TCRT5000. TCRT5000 assembled without a 4.7k resistor. Everything works fine (I will try to flash it on DualMoko).

It seems to me that 4.7 k is not needed.

Once again, forgive me if I was wasting your time and thank you very much for your library. You are my hero! :)

RyoKosaka, 2020-09-19:
Thanks for trying it out, I used a 4.7kΩ resistor on the first breadboard because I thought it could be used with the FSR and Roland controller when I eventually convert it to a PCB.
Regarding the MUX, try connecting the yellow signal wire to the pins of the MUX, just like the UNO.

NinjaPanas, 2020-09-19:
Thak you very much!

closed 2020-09-19 by NinjaPanas