RyoKosaka/HelloDrum-arduino-Library #27: Esp32

sgllab, 2020-10-24:
I'm having an issue with the velocity reading being inverted on an esp32. I added a quick velocity =1023-velocity in the single piezo function but I can't figure out what's doing it. I tried uncommenting the pull-up define but that wrecked more havoc so I think there's something somewhere doing it prior that I can't find. If you could point me what I'm doing wrong, that's help me a good bit. Thanks for all the hard work and help.

RyoKosaka, 2020-10-31:
Thanks for pointing that out. I tested it and reproduced the same phenomenon. I don't know where I went wrong, but I guess I didn't check it enough.

I'll fix it in the next release, but in the meantime, the following should work.

in hellodrum.cpp, line 144-146
`#ifdef ESP32
  piezoValue = 1023 - piezoValue / 4;
#endif`

Thank you.

closed 2020-10-31 by RyoKosaka

sgllab, 2020-11-02:
Thank you! Great programming btw. Saved me a ton of work. Best.

NinjaPanas, 2020-11-03:
> Thank you! Great programming btw. Saved me a ton of work. Best.

His work is just awesome, that's true. He is my hero! :)