RyoKosaka/HelloDrum-arduino-Library #29: Latency issue

rendel18, 2020-12-06:
Hi Sir,

I have latency issue using lcd shield kit and code for Smartwav2. But my other midi controller device has not latency even kat ktmp1 multipad to smartwav has no latency. Im using DIN 5pin midi adafter not usb midi. Thanks

RyoKosaka, 2020-12-10:
I'll be getting my smartWAV2 soon. Haven't been able to check it out yet.
If you send MIDI signals directly via the DIN5 pin, you do not need to set the baud rate.

Just
```cpp
    #include <MIDI.h>
    MIDI_CREATE_DEFAULT_INSTANCE();

    void setup()
    {
        MIDI.begin(); 
    }
    ```

rendel18, 2020-12-10:
Just only midi.begin() ? not midi.begin(10) or  serial.begin 31250 ?

RyoKosaka, 2020-12-10:
Yes.
MIDI.begin(10); 
This "10" is MIDI channel. And, if you not set this value, the MIDI signal is sent to all channel.
if you want to send single channel, use it.

and you dont need set Serial.begin(31250).
because MIDI library set baudrate 31250 inside library.

rendel18, 2020-12-10:
I hope you maximize the potential ability of smartwav2 and create a high
end module without pc. I have test it with nitro alesis module and its
works great. I insert sounds from ezdrummer 2 kits wav sound and import to
sd card. But one thing is left out. the choke and variable hihat cc also
dual and triple cymbals. I think thats the module midi command. If you
creat it using your arduino minimal kit its a big factor to all diy world.
Hope and best good luck. Thanks.

RyoKosaka, 2020-12-10:
Yes. I want to try it.
By the way, how did you export Wav files from EZdrummer2?

rendel18, 2020-12-10:
This trick is only allow to export sound wav in any drum software by using
SDSE software by lustark. Purchase it license to export fullkit sounds.
Check their website.

rendel18, 2021-03-11:
Hi Ryo,

How are you?

Do you have smartwav2 already?

rendel18, 2021-04-08:
hello sir

topogigio, 2024-04-22:
Hi Ryo,
I have 16ms latency with midi din using esp32 at 240mhz, any suggestions how to reduce it?
I also have enabled BLE and wi-fi (for parameter configuration, only with client connected) I try to disable BLE but latency is still the same.