RyoKosaka/HelloDrum-arduino-Library #30: Hi-Hat Pedal Nor working as it should

jhsa, 2020-12-06:
Hi, I am testing a pedal with the optical sensor TCRT5000 and EZdrummer 2.
When I open the pedal, it does show the pedal open on EZ2, but if I hit the piezo the pedal somehow half closes on the screen. Have you noticed something like this? Thanks for your help.

João

RyoKosaka, 2020-12-08:
Do you have E-drum Manufacture set up in Roland?
[image]

jhsa, 2020-12-09:
Hi. thanks for your reply. I didn't know we had to select a manufacturer. It is working now although it is a bit too sensitive. I am building a pedal right now. 
when i test it on the breadboard, moving a white card near the optical sensor, sometimes I get the sound of the closed hat at full blast. Perhaps too sensitive?

Thank you.

NinjaPanas, 2020-12-09:
> Hi. thanks for your reply. I didn't know we had to select a manufacturer. It is working now although it is a bit too sensitive. I am building a pedal right now.
> when i test it on the breadboard, [...]

Excuse me, your question is not for me.. but you should to see this - https://github.com/RyoKosaka/HelloDrum-arduino-Library/blob/master/docs/sensing.md#hihat-controller-sensing---hihatcontrol-hihatcontrolmux

Excuse me for my broken english. And excuse me if I give to you a wrong answer.

jhsa, 2020-12-09:
Any help is welcome. Thank you.

NinjaPanas, 2020-12-09:
> Any help is welcome. Thank you.

Thank you for your kindness!

So I think you can to adjust sensitive using "edit" button on device and that splendacious diagram with notes (on https://github.com/RyoKosaka/HelloDrum-arduino-Library/blob/master/docs/sensing.md#hihat-controller-sensing---hihatcontrol-hihatcontrolmux ).

jhsa, 2020-12-09:
Thanks.. I still have the arduino Nano on breadboard. I am waiting for some MUX chips.
At the moment, I am using one of the examples and am adding pads to it as I build them. This will give me a better understanding of the code O think.
Thanks

jhsa, 2020-12-09:
On the link from the post above, there is something that doesn't seem to agree with the example I am experimenting with. For the pedal options, it shows this:

- Sensitivity (1 - 100)
- Threshold (1 - 100)
- Scan Start (1 - 100)
- Scan End (1 - 100)
- Pedal Sensitivity (1 - 100)
- Velocity Curve Type (0 - 4)

But on the pedal example the last line is the pedal closed note, which is 44.

So, what is correct then?

jhsa, 2020-12-10:
Sorry guys, can't make this work properly. For example when I close the pedal completely, instead of a closed sound, I have an open sound. If I lift the pedal a bit, it does close. If i lift further, it starts opening again. I tried everything. Also, if I build the schematics of the TCRT5000 as per the instructions, the pedal works inverted. I have to collect the signal on the other side of the photo transistor for it to work in the right direction.

jhsa, 2020-12-10:
Something else is that I don't seem to be able to stop the HH close sound when I am moving the pedal. Tried all combinations of values. they don't seem to do much anyway.

RyoKosaka, 2020-12-10:
@jhsa 
As you can see in this video, the TCRT5000 cannot read values too close to the reflector, so the close position has to be about 5 mm away from the reflector.
https://www.youtube.com/watch?v=LbjRKrvGbME
As you point out, for some reason the values are larger if you are too close to the reflector. I'll add that to the README.

The HH Close sound is probably caused by the piezo in the HH, not the pedal, so please try raise the HH threshold.

What code are you testing with?

jhsa, 2020-12-10:
Thank you.. 
Yeah, I have measured the voltage on the sensor and saw that happening.. Now I lowered it a bit..
At the moment I only have the HH piezo on the table. I am breadboarding the circuit.. But I have the sensor on the pedal, not on the hi hat stand. The hi-hat will not move.

NinjaPanas, 2020-12-10:
> On the link from the post above, there is something that doesn't seem to agree with the example I am experimenting with. For the pedal options, it shows this:
>
> * Sensitivity (1 - 100)
> * Threshold (1 [...]

Well,  
'Pedal sensitivity' = 'Sensitivity' - 'Threshold'
'MIDI Velocity' = 'Pedal sensitivity' / 'Scan Time'
and 
'Scan Time' = 'Scan End' - 'Scan Start'

So you can change MIDI Velocity by few ways. If you need to reduce MIDI Velocity you can to increase 'Scan Time' or to reduce 'Pedal sensitivity'. 

I think that additional way is trying to change Velocity Curve Type. It is like type of potentiometer - Log, Lin and suchlike.

jhsa, 2020-12-10:
Yes, but as I said above, at least in the example, which is basically the same code used on the minimal DrumKit (I think it is called like this), does not have the Curve Type parameter.

I am also having trouble understanding what every parameter does. Perhaps because I am a guitar player  :) ;)
So, if I have  the HH close sound triggering when the pedal is moving between the two end points, how do I stop it?
As far as I know, the HH closing should only sound when I close the pedal quite fast, just like the real HH, right?
Thanks a lot for your help guys. I hope that this topic helps other people having the same problem.

jhsa, 2020-12-10:
When you say "MIDI Velocity" do you mean the Pedal CC number, or the Note Velocity of note 44 which is the pedal closing when you hit the pedal to the bottom?? This is what is confusing me.

RyoKosaka, 2020-12-10:
When you read the analogRead values on a serial plotter, do the values change smoothly? (If you put the reflector too close to the TCRT5000, it will be erratic.)
If it is changing smoothly then you should be able to get it to work correctly. What is the analogRead value when it is open and when it is closed?

 ```cpp
    void setup() {
        Serial.begin(38400);
    }

    void loop() {
        int piezoValue = analogRead(A0); //Connect the TCRT5000 to the A0 pin
        Serial.println(sensorValue);
    }
    ```

NinjaPanas, 2020-12-10:
>Yes, but as I said above, at least in the example, which is basically the same code used on the minimal DrumKit (I think it is called like this), does not have the Curve Type parameter.

Minimal DrumKit is simple. You shold to make it from https://github.com/RyoKosaka/HelloDrum-arduino-Library/tree/master/examples/Basic 
I can help with that! I'll try to compose code from examples to you - it is not very complicate. 

/*
  EXAMPLE - Simple Piezo Sensing
  With this sample code, you will make snare and tom using two piezo.
  https://github.com/RyoKosaka/HelloDrum-arduino-Library
*/

///////////////////////////////   SETTING VALUE   ///////////////////////////////////

//Determine the setting value.
//By changing the number in this array you can set sensitivity, threshold and so on.

byte HIHAT[6] = {
    100, //sensitivity
    10,  //threshold
    10,  //scan start
    30,  //scan end
    46,  //note of open
    42,  //note of close
};

byte HIHAT_PEDAL[6] = {
    90, //sensitivity
    30, //threshold
    60, //scan start
    80, //scan end
    10, //pedal sensitivity
    44  //note of pedal
};

byte SNARE[6] = {
    100, //sensitivity (1-100)
    10,  //threshold (1-100)
    10,  //scan time (1-)
    30,  //mask time (1-)
    38,  //note (0-127)
    1    //curve type (0-4)
};

byte TOM[6] = {
    100, //sensitivity (1-100)
    10,  //threshold (1-100)
    10,  //scan time (1-)
    30,  //mask time(1-)
    50,  //note (0-127)
    1    //curve type(0-4)
};

/////////////////////////////////////////////////////////////////////////////////////

#include <hellodrum.h>

//Using MIDI Library. If you want to use USB-MIDI, comment out the next two lines.
#include <MIDI.h>
MIDI_CREATE_DEFAULT_INSTANCE();

//Uncomment the next two lines for using USB-MIDI with atmega32u4 or Teensy
//#include <USB-MIDI.h>
//USBMIDI_CREATE_DEFAULT_INSTANCE();

//Please name your piezo.
//The piezo named snare is connected to the A0 pin and the piezo named tom is connected to the A1 pin.
HelloDrum hihat(0);
HelloDrum hihatPedal(1);
HelloDrum snare(2);
HelloDrum tom(3);

void setup()
{
  //If you use Hairless MIDI, you have to comment out the next line.
  MIDI.begin(10);

  //And uncomment the next two lines.
  //MIDI.begin();
  //Serial.begin(38400);

  //Set Curve Type
  snare.setCurve(SNARE[5]);
  tom.setCurve(TOM[5]);
}

void loop()
{

 //Piezo sensing is done in this line. And it is returned as a velocity of 127 stages.
  //For each pad, one line is required.
  //So, you need the same number of lines as the number of pads or controller.
  hihat.HH(HIHAT[0], HIHAT[1], HIHAT[2], HIHAT[3]);
  hihatPedal.hihatControl(HIHAT_PEDAL[0], HIHAT_PEDAL[1], HIHAT_PEDAL[2], HIHAT_PEDAL[3], HIHAT_PEDAL[4]); //hihatControl(byte sens, byte thre, byte scanStart, byte scanEnd, byte pedalSens);

  //MIDI signals are transmitted with this IF statement.
  if (hihat.hit == true)
  {
    //check open or close
    //1.open
    if (hihatPedal.openHH == true)
    {
      MIDI.sendNoteOn(HIHAT[4], hihat.velocity, 10); //(note, velocity, channel)
      MIDI.sendNoteOff(HIHAT[4], 0, 10);
    }
    //2.close
    else if (hihatPedal.closeHH == true)
    {
      MIDI.sendNoteOn(HIHAT[5], hihat.velocity, 10); //(note, velocity, channel)
      MIDI.sendNoteOff(HIHAT[5], 0, 10);
    }
  }

  //when pedal is closed
  if (hihatPedal.hit == true)
  {
    MIDI.sendNoteOn(HIHAT_PEDAL[5], hihatPedal.velocity, 10); //(note, velocity, channel)
    MIDI.sendNoteOff(HIHAT_PEDAL[5], 0, 10);
  }

  //sending state of pedal with controll change
  if (hihatPedal.moving == true)
  {
    MIDI.sendControlChange(4, hihatPedal.pedalCC, 10);
  }

  //Piezo sensing is done in this line. And it is returned as a velocity of 127 stages.
  //For each piezo, one line is required.
  snare.singlePiezo(SNARE[0], SNARE[1], SNARE[2], SNARE[3]);
  tom.singlePiezo(TOM[0], TOM[1], TOM[2], TOM[3]);

  //MIDI signals are transmitted with this IF statement.
  //For each piezo, one IF statement is required
  if (snare.hit == true)
  {
    MIDI.sendNoteOn(SNARE[4], snare.velocity, 10); //(note, velocity, channel)
    MIDI.sendNoteOff(SNARE[4], 0, 10);
  }

  if (tom.hit == true)
  {
    MIDI.sendNoteOn(TOM[4], tom.velocity, 10); //(note, velocity, channel)
    MIDI.sendNoteOff(TOM[4], 0, 10);
  }
}

(damn, on my old PC a can't to press "code" button. I hope it would be readable.)

NinjaPanas, 2020-12-10:
> When you say "MIDI Velocity" do you mean the Pedal CC number, or the Note Velocity of note 44 which is the pedal closing when you hit the pedal to the bottom?? This is what is confusing me.

Well - when HH pedal become closed - it make sound of HH. The MIDI Velocity is the Loudness of HH sound. When you closing your pedal slow enough than note sounds quiet.

jhsa, 2020-12-10:
Yes, that is thw code I am also using. I already have Kick, snare and side stick working quite well. My problem at the moment is only the Hi Hat pedal. And that is why I put thst code aside for now and loaded the Hi Hat pedal example, to make sure the other analog inputs are not interfering with it.
When I have all the single pieces working, I will try to put them all together with the LCD and everything. 
One thing that is helping a lot with calibrating is that I have placed potentiometers on every input (except the pedal) replacing the 1M resistor. I an using 500K pots but I think 250K is also ok.
This helps a lot with the sensitivity as it brings the sensor voltage to a usable range, before the code is applied..

jhsa, 2020-12-10:
I checked the pedal voltage with a multimeter and it seemed linear. Will try with the serial and analogRead. Thanks for the suggestion.

NinjaPanas, 2020-12-10:
> Yes, but as I said above, at least in the example, which is basically the same code used on the minimal DrumKit (I think it is called like this), does not have the Curve Type parameter.
>
> I am also [...]

Minimal Drumkit is simple. Try to compose your own code. 
`/*
  EXAMPLE - Simple Piezo Sensing
  With this sample code, you will make snare and tom using two piezo.
  https://github.com/RyoKosaka/HelloDrum-arduino-Library
*/

///////////////////////////////   SETTING VALUE   ///////////////////////////////////

//Determine the setting value.
//By changing the number in this array you can set sensitivity, threshold and so on.

byte HIHAT[6] = {
    100, //sensitivity
    10,  //threshold
    10,  //scan start
    30,  //scan end
    46,  //note of open
    42,  //note of close
};

byte HIHAT_PEDAL[6] = {
    90, //sensitivity
    30, //threshold
    60, //scan start
    80, //scan end
    10, //pedal sensitivity
    44  //note of pedal
};

byte SNARE[6] = {
    100, //sensitivity (1-100)
    10,  //threshold (1-100)
    10,  //scan time (1-)
    30,  //mask time (1-)
    38,  //note (0-127)
    1    //curve type (0-4)
};

byte TOM[6] = {
    100, //sensitivity (1-100)
    10,  //threshold (1-100)
    10,  //scan time (1-)
    30,  //mask time(1-)
    50,  //note (0-127)
    1    //curve type(0-4)
};

/////////////////////////////////////////////////////////////////////////////////////

#include <hellodrum.h>

//Using MIDI Library. If you want to use USB-MIDI, comment out the next two lines.
#include <MIDI.h>
MIDI_CREATE_DEFAULT_INSTANCE();

//Uncomment the next two lines for using USB-MIDI with atmega32u4 or Teensy
//#include <USB-MIDI.h>
//USBMIDI_CREATE_DEFAULT_INSTANCE();

//Please name your piezo.
//The piezo named snare is connected to the A0 pin and the piezo named tom is connected to the A1 pin.
HelloDrum hihat(0);
HelloDrum hihatPedal(1);
HelloDrum snare(2);
HelloDrum tom(3);

void setup()
{
  //If you use Hairless MIDI, you have to comment out the next line.
  MIDI.begin(10);

  //And uncomment the next two lines.
  //MIDI.begin();
  //Serial.begin(38400);

  //Set Curve Type
  snare.setCurve(SNARE[5]);
  tom.setCurve(TOM[5]);
}

void loop()
{

 //Piezo sensing is done in this line. And it is returned as a velocity of 127 stages.
  //For each pad, one line is required.
  //So, you need the same number of lines as the number of pads or controller.
  hihat.HH(HIHAT[0], HIHAT[1], HIHAT[2], HIHAT[3]);
  hihatPedal.hihatControl(HIHAT_PEDAL[0], HIHAT_PEDAL[1], HIHAT_PEDAL[2], HIHAT_PEDAL[3], HIHAT_PEDAL[4]); //hihatControl(byte sens, byte thre, byte scanStart, byte scanEnd, byte pedalSens);

  //MIDI signals are transmitted with this IF statement.
  if (hihat.hit == true)
  {
    //check open or close
    //1.open
    if (hihatPedal.openHH == true)
    {
      MIDI.sendNoteOn(HIHAT[4], hihat.velocity, 10); //(note, velocity, channel)
      MIDI.sendNoteOff(HIHAT[4], 0, 10);
    }
    //2.close
    else if (hihatPedal.closeHH == true)
    {
      MIDI.sendNoteOn(HIHAT[5], hihat.velocity, 10); //(note, velocity, channel)
      MIDI.sendNoteOff(HIHAT[5], 0, 10);
    }
  }

  //when pedal is closed
  if (hihatPedal.hit == true)
  {
    MIDI.sendNoteOn(HIHAT_PEDAL[5], hihatPedal.velocity, 10); //(note, velocity, channel)
    MIDI.sendNoteOff(HIHAT_PEDAL[5], 0, 10);
  }

  //sending state of pedal with controll change
  if (hihatPedal.moving == true)
  {
    MIDI.sendControlChange(4, hihatPedal.pedalCC, 10);
  }

  //Piezo sensing is done in this line. And it is returned as a velocity of 127 stages.
  //For each piezo, one line is required.
  snare.singlePiezo(SNARE[0], SNARE[1], SNARE[2], SNARE[3]);
  tom.singlePiezo(TOM[0], TOM[1], TOM[2], TOM[3]);

  //MIDI signals are transmitted with this IF statement.
  //For each piezo, one IF statement is required
  if (snare.hit == true)
  {
    MIDI.sendNoteOn(SNARE[4], snare.velocity, 10); //(note, velocity, channel)
    MIDI.sendNoteOff(SNARE[4], 0, 10);
  }

  if (tom.hit == true)
  {
    MIDI.sendNoteOn(TOM[4], tom.velocity, 10); //(note, velocity, channel)
    MIDI.sendNoteOff(TOM[4], 0, 10);
  }

}`

jhsa, 2020-12-10:
> Well - when HH pedal become closed - it make sound of HH. The MIDI Velocity is the Loudness of HH sound. When you closing your pedal slow enough than note sounds quiet.

Well, this is the problem, sometimes it sounds when I am moving the pedal. It shouldn't.

NinjaPanas, 2020-12-10:
@ #

>
[code.txt](https://github.com/RyoKosaka/HelloDrum-arduino-Library/files/5672213/code.txt)

My code example

NinjaPanas, 2020-12-10:
> Well, this is the problem, sometimes it sounds when I am moving the pedal. It shouldn't.

Try to change - Velocity Curve Type - for HH pedal

RyoKosaka, 2020-12-10:
@jhsa 
Are you testing with Example > Basic > hitat.ino or hihhat_2zone.ino? 
As you say, this sample is old and unfriendly without any curve settings.
I'll try to fix it a bit.

jhsa, 2020-12-10:
> Try to change - Velocity Curve Type - for HH pedal

There isno Curve Type :)

jhsa, 2020-12-10:
> @jhsa
> Are you testing with Example > Basic > hitat.ino or hihhat_2zone.ino?
> As you say, this sample is old and unfriendly without any curve settings.
> I'll try to fix it a bit.

I am using the basic example with pedal. Will let you know which one in a while.. computer is still off :)

RyoKosaka, 2020-12-10:
I've updated the Example a bit and added some comments. Please check it out.
https://github.com/RyoKosaka/HelloDrum-arduino-Library/blob/master/examples/Basic/hihat/hihat.ino

jhsa, 2020-12-10:
I will now thank you.. 
Also, I think the problem is that I had to wire the sensor differently from your schematics for it rto work in the right direction. You have the resistors and the analog pin connection on the positive side of the sensor.. I had to connect them to the negative side for the direction to be correct.. Or am i missing something?

Thanks

jhsa, 2020-12-10:
On you last example there is a ";" missing on line 63.
It won't compile without it obviously :)

jhsa, 2020-12-10:
with your last code, it seems that my pedal now is reversed. :(

jhsa, 2020-12-10:
10, //sensitivity (Hold the pedal at the position where you want to send **127(full open)** in the MIDI CC, and read the Analog Read value in the TCRT5000. Divide the value by 10 and enter the va

70, //scan end (When the analogRead value reaches this value * 10, the scan is finished and the time taken for the scan is referenced to determine the velocity.Then noteOn will be sent.The default value is 90, the same as Sensitivity. **This means that the pedal is closed at the point where MIDI CC is 127.)**

This doesn't make sense. According to your text, pedal should be fully open at 127. But then later you say that the pedal should be closed at CC127? Have a read at your text :)

NinjaPanas, 2020-12-10:
> This doesn't make sense. According to your text, pedal should be fully open at 127. But then later you say that the pedal should be closed at CC127? Have a read at your text :)
 Can be slight inaccuracy in comments. What about new code, is it working for you?

jhsa, 2020-12-10:
Nope.. I now connected as sensor as the instructions suggest. Analog input connected to the positive side.. Pedal movement is correct, but pedal closed is reversed.. It closes when i actually lift the pedal.

RyoKosaka, 2020-12-10:
Oops. It's the other way around. I haven't touched it in a while, so I forgot.
I don't have an Arduino at hand, so I'll try and fix it later.

You already know how it works, right?
 If the hi-hat is moving in reverse, just swap the threshold and sensitivity values. The same goes for scanning.

jhsa, 2020-12-10:
Hmm, I try again, but I think I did and it didn't work.. let me try again.. 
As far as understanding, I think Threshold and sensitivity are the end points.. The others I still have to understand. I do apologize for being stupid.. But this looks like a really cool project, and your code is the one that still worked the best for me.. So, i am not giving up just yet.

RyoKosaka, 2020-12-10:
Try commenting setCurve(0). That might be the cause...
Anyway, the update I just made is a lazy one, and I'll examine it carefully later.

It's because of people like you who are willing to try that I can continue to develop this project and improve it, slowly and gradually. Thanks for giving it a try. :)

jhsa, 2020-12-10:
And I say Thank You to both of you.. Thank you so much.

NinjaPanas, 2020-12-10:
> And I say Thank You to both of you.. Thank you so much.

no, no, no - I'm just a (just not quite smart enough) user of HelloDrum with broken english. All thank 2 RyoKosaka for his lib. 

I tried different libraries (but mostly midi controllers for arduino, not drums) from Internet and this is the best of the best. I mean - it is quite complicated things with all this "boar type" (if you need dualMoco), "wiring diagrams", LCDs... And all the things you can find using "https://github.com/RyoKosaka". Mostly it is https://github.com/RyoKosaka/HelloDrum-arduino-Library , but some time I look to https://github.com/RyoKosaka/drums .

And after all - this feature with EEPROM is exactly what I need so that I can configure the device in the "field" without PC. I think it's great. And it is only one.

jhsa, 2020-12-10:
Well you have been trying to help me, so I thank you too :)

jhsa, 2020-12-10:
If I have the resistors on the ground side, then closing is not reversed but it is sending the pedal closed note  when the pedal is open and it just started to close. It should send it when it is nearly closed.

jhsa, 2020-12-10:
These are my values. 

byte HIHAT_PEDAL[7] = {
    99, //sensitivity (Hold the pedal at the position where you want to send 127(full open) in the MIDI CC, and read the Analog Read value in the TCRT5000. Divide the value by 10 and enter the value.)
    73, //threshold (Hold the pedal at the position where you want to send 0(close) in the MIDI CC, and read the Analog Read value in the TCRT5000. Divide the value by 10 and enter the value.)
    70, //scan start (When the analogRead value reaches this value * 10, the scan will start and the time will be recorded.)
    95, //scan end (When the analogRead value reaches this value * 10, the scan is finished and the time taken for the scan is referenced to determine the velocity.Then noteOn will be sent.The default value is 90, the same as Sensitivity. This means that the pedal is closed at the point where MIDI CC is 127.)
    10, //pedal sensitivity (This value multiplied by 100 is the time used to determine the velocity of the pedal. If the time taken from the scan start position to the scan end position is more than 1000 milliseconds, the velocity will be set to 1. So, if you feel that the velocity is low, either increase pedalSensitivity or increase the value of scan start.)
    1,  //curve type
    44  //note of pedal

jhsa, 2020-12-10:
Also, if i place the pedal on the exact position that the close HH note is sent, the note is sent continuously. Perhaps some averaging is needed??

jhsa, 2020-12-10:
Ok. Managed to make it work a little bit better. but it is still not 100%. Sometimes the pedal closing sound still plays when it shouldn't, and also sometimes the sound of the pedal opening after you closed it too hard.

jhsa, 2020-12-11:
byte HIHAT_PEDAL[7] = {
99, //sensitivity - This is the lower Pedal Range limit. when it is fully closed
73, //threshold - This is the upper Pedal Range limit. when it is fully open
73 //scan start - Beginning of scan for the Pedal Closing sound velocity number. it starts counting when the pedal is fully open.
80, //scan end - this Plays the closing sound. In this case just after the pedal was moved from open, with a velocity depending on the time since Scan Start.
15, //pedal sensitivity
2, //curve type
44 //note of pedal

Well, did I finally understand what the parameters do)) ;) 
Still, I think averaging is really needed when the pedal plays the closing sound.

jhsa, 2020-12-13:
Hi guys, I more or less solved that pedal problem.. well, it is about 90% solved :) 
Wouldn't it be a good idea to make some kind of forum where we can discuss the project and help each other? Perhaps a Facebook group or something?
I am waiting for some MUX ICs so i can add more pads. I am sure i will need help with it, specially with those 2 and 3 zone cymbals. I will for example need choke on the crash cymbals :)
Here is a little video showing a bit of my progress. You can see the pedal operation.. Attention, I am not a drummer :)

https://www.youtube.com/watch?v=l6050aAb9yA

RyoKosaka, 2020-12-14:
This is a great job. Thanks for sharing the video!!
I think creating a forum is a good idea. I thought about making a wiki with documentation and videos on how to set it up, but it's a lot of work and I haven't gotten around to it.

jhsa, 2020-12-14:
If you are on Facebook, you could create a group there. Or if you want I can create one.

RyoKosaka, 2020-12-14:
Could you please create a group? Thanks.

jhsa, 2020-12-14:
No.problem. will do..

jhsa, 2020-12-15:
Ok, I have created a Facebook group. Here it is:

https://www.facebook.com/groups/838951150235243

It is a private group.
RyoKosaka, please join the group. You will have to answer 2 simple questions and agree to the group rules to become accepted as a member. After you joined it, I will make you an administrator. Is that ok?

RyoKosaka, 2020-12-15:
OK, thank you so much!

closed 2020-12-15 by RyoKosaka