RyoKosaka/HelloDrum-arduino-Library #31: Send multiple notes

TomMaradan, 2021-01-08:
hello
I have a problem with my setup
When I hit my pad made with a diy trigger, wire on one analog pin, the arduino send two or more midi notes and ezdrummer hit many instrument in the same time. 
However some works well, I noticed that it was from 3 connected element, that the 3rd had this problem

Could you please help me?

Thank you 
Tom maradan

(I'm french, sorry for mistakes)

assenheimer, 2021-09-09:
Hi!
Check the sensibility and threshold configuration. And check the construction of your pads too.
I have similar issues, one pad vibrates the piezo from another pad when i hit hard.

(I'm brazilian, sorry for mistakes too)

DavosPegas, 2022-02-03:
Same happens for me. It's definitely  not only pads' constructional or threshold settings issue as one of notes that are triggered randomly is kick, which is separated from the rest of octapad (kick pedal on 1m long cable). I expect it may be connected to a) Arduino  possibilites or b) some electrical stuff (wiring or resistor or smth)...

DavosPegas, 2022-02-05:
Ok, to be honest, I've put some efforts here. Indeed this problem is solved for me only by proper settings. I changed thereshold somwhere to 20-40 and Sensitivity to 60-100 for all pads and it works muuuch better. Although quite soft playing is necessary.

fedetarrats1, 2023-04-23:
Hey guys, did you ever find a solution for this? I'm building a big drum set and I can hit one pad quickly without any problem but as soon as I start hitting 2 at the same time one of them stops sending the signal, I tried different combinations of pads with the same result

marcospozzo, 2023-04-27:
Hi! I am having the same issue. When I hit the crash cymbal (midi note 49), the kick (midi note 36) also triggers. Kick is on A0 and Crash on A5 on an Arduino Uno board. Thanks in advance!

ardumidi, 2023-11-03:
Alguma solução para isso?

zhdan88vadim, 2025-01-24:
Hello
I think this may be related to the features of the microcontroller.

Because the ADC of microcontroller switches channels, and it is possible that at some point in time during switching the signal remained at a high level.

These are just my thoughts.

[image]

source article:
https://deepbluembedded.com/arduino-adc-analogread-analog-input/

There are also other electronic drum projects. It is necessary to study their implementation and whether they have the same problems.

zhdan88vadim, 2025-01-24:
perhaps an additional signal pull to the ground would increase the signal attenuation rate. 
I have also seen more complex signal amplification circuits on operational amplifiers. I like this option more, then you can play with high sensitivity of the pads.

Professional advice is needed here.

zhdan88vadim, 2025-01-29:
Today I lengthened the wires to the sensors and the number of false positives has increased. The problem seems to be the shielding of the signal wires.

zhdan88vadim, 2025-01-29:
I found a solution to my problem. Using this code, I watched the signals from my sensors. And when I take the sensor in my hands and hit it, there is no effect on the neighboring sensors. **The problem was in poor vibration isolation.**

`
void setup() {
  Serial.begin(115200);
  delay(1000);
}

void loop() {
  int a0 = analogRead(A0);
  int a1 = analogRead(A1);
  int a2 = analogRead(A2);
  int a3 = analogRead(A3);
  int a6 = analogRead(A6);
  int a7 = analogRead(A7);

  Serial.print(a0);
  Serial.print(",");
   Serial.print(a1);
  Serial.print(",");
  Serial.print(a2);
  Serial.print(",");
  Serial.print(a3);
  Serial.print(",");
  Serial.print(a6);
  Serial.print(",");
  Serial.println(a7);

  //delay(1);
}`

[image]

In the picture you can see how one sensor influences the second one through vibration