RyoKosaka/HelloDrum-arduino-Library #32: is analog megadrum board compatible?

lmeucchi, 2021-03-26:
Hi how are you?
I have made the analogue megadrum board and I want to use it to make a HelloDrum with 32 inputs. 
In a diagram that you published "Arduino UNO with MUX" I read that it is based on that board, but it does not use the same values ​​of resistance and connections.
What should I do to make that board work with Arduino?
Thanks greetings
[image]

RyoKosaka, 2021-03-27:
I once connected a circuit almost identical to the megadrum schematic to an Arduino, but I couldn't get it to work well with helloDrum, so I modified some parts of it.
I tried it a long time ago, so I don't remember what went wrong. It might be worth trying again.

jmartens2, 2021-05-17:
i would also ask if this could be ported to operate a megadrum. ive lost my working firmware due to hardware failures. ive tried to resurrect the original project but too many compile issues. thank you for your time if youve read this! great work. have a great week.