RyoKosaka/HelloDrum-arduino-Library #38: Mux or no mux?

meekowai, 2021-07-21:
Hi.
Can you explain what is the difference between verions with MUX and without? I'm new to this. Thanks.

assenheimer, 2021-09-09:
Hi!
MUX version is used in case you have a mux shield to expand the analogues ports of arduino

ardumidi, 2023-11-03:
Olá, estou tentando usar a sua biblioteca para fazer uma bateria com arduíno Leonardo usando multiplexador 4067, porém mesmo com o exemplo pronto da biblioteca parece que o mux não está trabalhando corretamente.

Quando vc toca em um pad, o arduíno envia duas notas diferentes as vezes até 3.

Testei diversas vezes meu circuito com outros códigos que tenho e utilizando o mux, e todos funcionam corretamente.

Porém quando carrego o exemplo e deixo apenas os 9 pads, uns funcionam bem, outros enviam duas notas, se conseguir me ajudar ficarei grato!