RyoKosaka/HelloDrum-arduino-Library #39: Simplified lib

DavosPegas, 2022-01-08:
Hi Ryo, I am making octapad with custom menu and some leds. Due to easy USB-MIDI communication i chose Arduino Micro. Now, the problem I have is available space. I've reached 80% of available space for variables and in next stęp will be working on some code optimization but wanted also to ask you:

Have you thought about light version of Hello Drum? No EEPROM or Settings or other fancy features, just rough piezo sensing and mux support. 

Can you prepare simplified version?

RyoKosaka, 2022-01-14:
Hi.
That is an interesting suggestion.
Certainly, this library is not suitable for Arduino with small capacity.

I'm not sure how lightweight it can be, but I'll give it a try.

DavosPegas, 2022-01-14:
Glad to hear that. Please drop a message here when it will be ready so I can be informed via email notification. Then I'll give it try. Happy to share how much space it will save in my case and if it works correctly.

RockVicedomini, 2023-07-13:
Any progress on this issue? Unfortunately including the U8g2lib library togheter with HelloDrum to show basic info on an OLED display seems not possible. 
I made a test with the code "muxSensing_U8g2.ino" included in the examples/MUX folder adapted to include USB-Midi as well and the IDE fails to compile providing the message:
_"Global variables use 2731 bytes (106%) of dynamic memory, leaving -171 bytes for local variables. Maximum is 2560 bytes"_  

Any useful hint?

Thank you anyway for your great job!