RyoKosaka/HelloDrum-arduino-Library #40: Inquiry

JerryDaga, 2022-02-03:
This code is compatible with all types of Arduino boards?
Mine is Arduino Leonardo and I wanna make my own D.I.Y. Table Top E-Drums Set.

DavosPegas, 2022-02-03:
I did octapad with Pro Micro (in IDE it is marked as Leonardo), so it should work for you as well.

JerryDaga, 2022-02-03:
Do you have a link for Arduino Leonardo? Sent from Mail for Windows 10 From: DavosPegasSent: Thursday, 3 February 2022 11:49 pmTo: RyoKosaka/HelloDrum-arduino-LibraryCc: JERRY OCAN; AuthorSubject: Re: [RyoKosaka/HelloDrum-arduino-Library] Inquiry (Issue #40) I did octapad with Pro Micro (in IDE it is marked as Leonardo), so it should work for you as well.

JerryDaga, 2022-02-05:
> I did octapad with Pro Micro (in IDE it is marked as Leonardo), so it should work for you as well.

This code for Arduino Leonardo can be used as controller for the emulation and other rhythm games?