RyoKosaka/HelloDrum-arduino-Library #42: There is no note hit, only "pitch bend"

kagandary, 2022-02-07:
Hey everyone, recently i been trying to make my own 14 channel drumset, i taught that it won't be that hard but when i completed circuit, installed the software to arduino i faced with the problem that only "pitch bend" parameter was changing. My circuit was including mux, and i taught that it was the problem but when i tried the most basic circuit in that project ( https://github.com/RyoKosaka/HelloDrum-arduino-Library/blob/master/examples/Basic/SimpleSensing/SimpleSensing.ino )i get that problem again.

Does that problem happens because of HelloDrum library? even i dont change anything (i tried the otherwise too) i still get only one knob changing. Can anybody tell me what im doing wrong so i can solve that?

[image]

RyoKosaka, 2022-02-07:
Perhaps it is the baud rate setting.
Are you specifying a baud rate of 38400 in the HairessMIDI settings?

Have you changed the code as follows?
```
//If you use Hairless MIDI, you have to comment out the next line.
//MIDI.begin(10);

//And uncomment the next two lines.
MIDI.begin();
Serial.begin(38400);
```

kagandary, 2022-02-08:
> Perhaps it is the baud rate setting. Are you specifying a baud rate of 38400 in the HairessMIDI settings?
>
> Have you changed the code as follows?
>
> ```
> //If you use Hairless MIDI, you have to comment [...]

i changed baud rate and now im getting right notes, thank you so much for your solution and quick reply