RyoKosaka/HelloDrum-arduino-Library #43: 3 piezo, hit Position sending

SunboX, 2022-04-23:
Is it possible to have three piezo per drum and calculate the hit position using triangulation?

corrados, 2022-04-24:
What do you want to measure? A time difference between the detected peaks of the three piezos? Let's assume sound travels at about 30000 cm/s. Then the time difference between peaks if the distance is 10 cm would be about 3 us. This would mean you have to sample at least with 300 kHz.

I think it only would be feasible to use the peak amplitude of the three piezos and maybe also the shape of the mean peak to detect the correct position on the mesh head.

SunboX, 2022-04-24:
My idea was to measure the amplitude of all three piezo. My guess is, that it is different depending where you hit the mesh. If so, you could use this information to calculate how far it was hit from the center.

Seems I have to make a test.

ojungleboy, 2022-04-27:
Totally support this idea!