RyoKosaka/HelloDrum-arduino-Library #48: Resistor and Zener values for ESP32 (3.3v boards)

ojungleboy, 2022-07-03:
First, thanks for your job.

Question
1M and 5.1v are the choice for 5v boards for resistor and diode respectivelly.

In case of 3.3v logic boards, what are the resistor and diode values for protection?

Thank you so much again!

prostealth, 2022-07-11:
if what you mean is for the piezo, you can use 1N4148 diode and 1M resistor will work too.

ojungleboy, 2022-07-11:
> if what you mean is for the piezo, you can use 1N4148 diode and 1M resistor will work too.

Thank you for your reply.

Yeah, I mean the piezo circuit. All diagrams using Arduino 5v is using one circuit. I am interested in the circuit for the ESP32 and I guess due to the lower logic level on the board the piezo circuit for ESP32 had a different circuit components for 3.3v

prostealth, 2022-07-12:
there are even a piezo circuit with only 1M resistor. as remember it was to prevent analog signal floating (hitting itself in case of arduino drum). the type of diode doesnt really matter. but its better to using the diode

closed 2024-02-28 by ojungleboy