RyoKosaka/HelloDrum-arduino-Library #59: Hello, I am having the same trouble.

ojungleboy, 2024-02-26:
I am trying to use MUX with ESP32, The baud rate is matching to the Hairless serial but it seems that the MIDI messages are coming with garbage
[image]

Both 4051 and 4067 MUXs are presenting the same issue.

Does anyone know how to solve it?

Thank you

_Originally posted by @ojungleboy in https://github.com/RyoKosaka/HelloDrum-arduino-Library/issues/51#issuecomment-1962800891_

ojungleboy, 2024-02-28:
If someone could help troubleshoot this would be nice. Thank you so much

steeltape, 2024-09-06:
I have the same problem.. did you figured out how to fix?

ojungleboy, 2024-09-06:
I kinda did solve it. There is something bugged in the examples related to
MUX code. I guess it it's regarding the velocity curve array.

The way I did was to get a simple piezo code and gradually adding the
second piezo code, and finally addind the other piezos. Finally I added the
mux state and included one by one at the MUX pins and it worked. I am not
on my pc right now, but have a shot on this approach and let us know

- Pedro Zinn Hensel Nunes

steeltape, 2024-09-06:
> I kinda did solve it. There is something bugged in the examples related to MUX code. I guess it it's regarding the velocity curve array. The way I did was to get a simple piezo code and gradually [...]

if you could share your code is much appreciated

ojungleboy, 2024-09-06:
Dear friend, I have no access to the code right now. On the pad declaration
try to check the order of parameters of the pad

- Pedro Zinn Hensel Nunes